rosco-m68k/rosco_m68k #92: Firmware should only optionally use GPIOs

roscopeco, 2020-08-18:
Currently, if the timer tick is enabled, it will always toggle GPIOs (to flash the green LED). This limits the utility of the GPIO port and makes efficient low-level programming more difficult.

While one can disable the timer completely with `bclr.b  #5,MFP_IMRB`, this also disables updating of the uptime counters, and so is not optimal. There should be a variable in the SDB that one can set to disable the GPIO toggling, while still allowing the timer to continue operating.

XarkLabs, 2020-08-19:
Thinking a little bit about this, my proposal would be to have a documented "GPIO mask" that has bits set that are "user use" that the system should not alter (default ~0x83 = LEDs, and CTS - and as discussed red LED will always blink on "crash" regardless).    Then the few places where firmware etc. touches GPIO change bset/cclr/bchg test the mask (or probably better) just read/modify/write byte using gpio_mask. This also functions as a simple "indicator" as to what GPIOs are in use (which might be handy).

roscopeco, 2020-08-19:
I like this idea @XarkLabs , seems more flexible than a simple Boolean. Good call :)

commit a7747ee8 (2020-11-26): Merge pull request #151 from rosco-m68k/feature/fix-92

Fix #92
commit cb3196d3 (2020-11-26): Fix #92
closed 2020-11-26 by XarkLabs