rosco-m68k/rosco_m68k #112: RS232-USB PL2302 causes problems after loading

nbrok, 2020-09-20:
I use no FTDI cable but an RS232 to USB cable (PL2302). With this I have an issue that after loading a program via Kermit that no ready message is shown. This due an USB reset after Kermit did load the file.
As I insert a readchar call in kmain of ehbasic or adventure then I can get the start-up message of that particular program after pressing a key. This is a dirty hack. I think the delay after loading a program is to short. (So I read from the ROM sourcecode)

nbrok, 2020-09-21:
I solved it by changing the delay value in main2.c of the firmware.
OLD:
// Wait a short while for the user's terminal to come back...
    mcBusywait(100000);
NEW:
// Wait a short while for the user's terminal to come back...
    mcBusywait(400000);

This results in 720 ms delay at 10MHz

roscopeco, 2020-09-22:
Reopening to cover work to make this happen. 

Increased delay should be applied as a hot fix to 1.1, 1.2 and also on Develop.

commit 8a727218 (2020-10-04): Merge pull request #131 from rosco-m68k/feature/fix-112

Fix #112
commit a108fde5 (2020-10-04): Fix #112
closed 2020-10-04 by XarkLabs