rosco-m68k/rosco_m68k #175: Toolchain install documentation possibly incomplete?

bzuidgeest, 2020-12-26:
I wanted to look around this project (at the moment your store is unfortunately closed) and tried following the tool-chain install documentation on 

https://rosco-m68k.com/docs/toolchain-installation

However for my linux mint install this proved incomplete. The one-liner for installing brew works, but afterwards I could not use brew. 
The brew documentation on https://docs.brew.sh/Homebrew-on-Linux lists 4 additional lines for getting brew added to the path. However even that fails to mention that you need to log off and on again for it to work. 

Maybe this is just something with my linux mint install, but it is pretty much default. So I thought I mention it here.

bzuidgeest, 2020-12-26:
On https://rosco-m68k.com/docs/building-the-software i found another error. 
Both git clone commands did not work on my linux machine (git clone github.com/rosco-m68k/rosco_m68k.git)
They needed to be changed to git clone *https://*github.com/rosco-m68k/rosco_m68k.git**

roscopeco, 2020-12-30:
Thanks for the report @bzuidgeest , will look to get this fixed as soon as possible :)

roscopeco, 2021-11-12:
(Finally) fixed. See:

* https://rosco-m68k.com/docs/toolchain-installation
* https://rosco-m68k.com/docs/building-the-software

Apologies for the long lead time on this, thanks again for reporting @bzuidgeest .

closed 2021-11-12 by roscopeco