rosco-m68k/rosco_m68k #176: VASM sha256 hash mismatch.

bzuidgeest, 2020-12-26:
When brewing I got an error when getting to vasm-all

Error: SHA256 mismatch
Expected: 2b7aba9b6d0a196a2ab009fbed08f10acd94da41d11d3a224cb59b2a6c2f2b41
  Actual: 9ae0b37bca11cae5cf00e4d47e7225737bdaec4028e4db2a501b4eca7df8639d
    File: /home/****/.cache/Homebrew/downloads/31e078af95a73fde88e3d55dc37179f5520233a98b3f351a785b0a3f9f54f314--vasm.tar.gz
To retry an incomplete download, remove the file above.

I looked into the formulae and it links to sun.hasenbraten.de/vasm/release/vasm.tar.gz as the download url.
Version is set at 1.8f. So I assume the sha256 is for that version. However when visiting page
http://sun.hasenbraten.de/vasm/index.php?view=relsrc which lists this url, it notes that the current version is v1.8i
So I assume the formulae or however homebrew calls its scripts is out of date.

VASM also has a page with tagged code releases http://sun.hasenbraten.de/vasm/index.php?view=tagged

there you can find links to a specific source release version. This seems a more stable download link to me for use with the formulae. So my advise would be to switch the link to using that (http://phoenix.owl.de/tags/vasm1_8f.tar.gz in this case). Otherwise you will always have it breaking on new releases of vasm. Maybe there are other options as I am no expert on this.

roscopeco, 2020-12-30:
Thanks for the report @bzuidgeest , I was not aware of the tagged releases. I have raised https://github.com/rosco-m68k/homebrew-toolchain/issues/7 in the toolchain repo (linked to this issue) to fix this. 

Thanks again! :)

roscopeco, 2021-01-18:
Closed, thanks @0xTJ :)

closed 2021-01-18 by roscopeco