rosco-m68k/rosco_m68k #178: Kermit problems

bzuidgeest, 2021-02-06:
I finished building my rosco-m68k and I am trying to follow the instructions on downloading one of the examples.

I says I need minicom and c-kermit. My linux mint install only has minicom and gkermit. minicom works and I get the ascii art, but gkermit does not seem to work. I cannot find anything on how to set ROBUST ans such settings for gkermit

I tried using brew to create c-kermit. But the brew recipe does a "make macos" which obviously does not work on linux. 

I then tried compiling from source using stable c-kermit from http://www.kermitproject.org. That would not compile either. After some research I found that it is incompatible with glibc2.28 or higher. The latest alpha version (9.0.305) which is about 10 years newer does compile.

I think for linux users the manual needs some "upgrading" Either it needs instructions for gkermit or a note on where to get a c-kermit version that compiles on modern linux systems. (a different transfer protocol would also be an option, but that seems rather involved)

roscopeco, 2021-03-04:
Thanks for reporting @bzuidgeest , will look to get the documentation upgraded in this area :)

roscopeco, 2022-01-14:
Closing as this issue is now documented under https://rosco-m68k.com/docs/connecting with a link to the rosco_m68k Kermit fork (https://github.com/rosco-m68k/ckermit).

closed 2022-01-14 by roscopeco