rosco-m68k/rosco_m68k #303: Warm boot doesn't re-initialise DUART after hardware reset

0xTJ, 2022-01-25:
Currently, the line https://github.com/rosco-m68k/rosco_m68k/blob/d2e599dbe13a67ec5b3cbc58612797911bdc1022/code/firmware/rosco_m68k_v2.1/main1.c#L116 has the warm boot handler iset to be used both at program exit and as hardware reset vector. This causes problems when using the system `/RESET` signal to reset the processor, because this also resets the DUART (and could reset any future added peripherals). The warm boot isn't made to re-setup the entire system, just to call the current program loader, so the warm boot message isn't printed, and Kermit doesn't work.

I'd propose that the hardware reset vector use a separate routine that's one step "colder" than the current warm boot.

It might also be worth considering splitting out some of the code in `bootstrap.asm` into separate files to make things a little more modular, but that's a separate issue.

roscopeco, 2022-01-25:
Interesting. This makes sense that it should be a problem, but I don't see it in (e.g. `updateflash` which uses the `reset` instruction and AIUI _should_ trigger this issue, but doesn't, at least on my system) 🤔 

Is there a test we can run which exhibits this behaviour?

(And yeah, I agree on `bootstrap.asm`. Breaking it up a bit has been on my list for quite a while now...)

roscopeco, 2022-01-25:
Having drunk plenty of coffee and thought more about this, I think it's good as currently implemented. I kind of feel like, if a piece of code is going to run a `reset` instruction (which it's worth noting doesn't actually reset the processor), then it's not really doing a warm reboot, and has no business expecting to be able to jump to the warm boot vector, as it has deliberately reset all the peripherals.

I think where `reset` is used, it's a reasonable requirement that the next thing the software (or firmware, I'm not against adding a trap for this) does is load the stack and initial PC from the ROM base and jumps there for a "cold" reboot.

What do you think?

roscopeco, 2022-01-25:
Ah right, okay, I can see your use-case now. 

I'm not _against_ the above in principle (but I do want to think it through more and digest it somewhat), and you're right that it's not technically supported with the current expansion bus (though likely just yanking `/RESET` will "work fine" with any 68010 you care to try it with). If you have chance to work something up I'd gladly look at it 😃 

FWIW I _do_ think we can achieve broadly what you need with a change in the glue logic. Specifically, if you replace IC5 with:

```
GAL22V10
GLUE

AS   IOSEL   HWRST  FC2    FC1     FC0   A1    A2    A3     A19      NC     GND
NC   NC      HALT   RESET  RUNLED  BQ0   BQ1   BQ2   BOOT   ANYIACK  DUIACK VCC

HALT.T  =  GND
HALT.E  =  HWRST

RESET.T =  GND
RESET.E =  HWRST

RUNLED  = /HALT

AR      = /RESET
BQ0.R   =  RESET                ; High after one memory cycle
BQ1.R   =  RESET * BQ0          ; High after two memory cycles
BQ2.R   =  RESET * BQ1          ; High after three memory cycles
BOOT.R  =  RESET * BQ2          ; High after four memory cycles

/ANYIACK=  RESET * FC0 * FC1 * FC2 * A19
/DUIACK =  RESET * FC0 * FC1 * FC2 * /AS * /A1 * /A2 * A3 * A19

DESCRIPTION

Modified IC5 for #303 test purposes
```

I _think_ that'll work, and should still be safe for the software reset case, but I'll admit I've not thought it through in depth or done any testing with it.

roscopeco, 2022-01-25:
(And actually, thinking a bit more about it, whether we change the firmware in support of this or not, we might want to think about using some slightly more robust version of the above for IC5 if we want to support hardware-reset-other-than-by-the-button. Off the top of my head we could use a spare pin to feedback a combination of `/HALT` and `/RESET` and use that to reset the `/BOOT` line maybe 🤔 ).

0xTJ, 2022-01-27:
Hmm, I'm not sure if that would work properly, since the processor can still do a system reset via `/RESET`, but continue executing. I imagine if would mess with things up if `/BOOT` got asserted at a time other than power-up reset.

I'm not sure how it'd be possible to differentiate an external `/RESET`-only reset from a `reset` instruction.

roscopeco, 2022-01-27:
Yeah, for the latter comment I'm thinking we should probably feed back `/RESET` and `/HALT` and reset `/BOOT` only if _both_ go low (i.e. a 'proper' hardware reset) rather than doing with the `HWRST` line as currently (it would be broadly the same thing, except it would support external reset). This would be more for completeness than anything else (as you say, it's not currently easy to do, as `/HALT` isn't broken out to the expansion bus).

(WRT the former comment, I just did a quick test, and the board continues to work with that change, which suggests there's likely more to it than met the eye when I came up with it - as you say, it really shouldn't work 🤔 I don't have time to get the LA on it right now, but it wasn't really a suggestion of "something we should do" anyway 😅 ).

0xTJ, 2022-01-28:
Ah, I see what you're saying, that makes sense and should be easy.

I'll make undo some of the changes to my prototype branch given that all external resets will be resets with to ROM with `/BOOT`, and I'll be making  PR since that makes it easier to what's changed. Though, with that PLD change, some of that I've said before wouldn't apply, so it's going to turn more into a `bootstrap.asm` refactor, and making things a little more modular with initialization.

The main part of what I said before that's more of a design decision to make is whether or not the reset vectors are still replaced with values for warm boot, since those wouldn't be "naturally" reachable anyways if all processor resets go to ROM. (I changed those to stay as a mostly-cold reset.)

PR #306 (2022-02-02): External reset will enable `BOOT` line reset from ROM
This is a slight variant of the change proposed in https://github.com/rosco-m68k/rosco_m68k/issues/303#issuecomment-1021571118. Performing an external processor reset will enable the `BOOT` line, with countdown.

Initially, I used the one unused output as a feedback of `HWRST + /HALT * /RESET` as `DOBOOT`. This then took the place of `HWRST` in the registered equations. Afterwards, I noticed that I could keep the pin free by just using `/HALT` and `/RESET` directly, since `HWRST` drives those low.

This fixes #303, since external reset now  [... cut]
closed 2022-02-02 by 0xTJ