rosco-m68k/rosco_m68k #315: Memcheck hangs at start of roscram_68k expansion ram board mapped memory space on rev 2.12 board

DrItanium, 2022-04-01:
Observed with a rosco_m68k board revision 2.12 with firmware 2.16 and a roscram_68k board. 

Regardless of the amount of ram installed in the expansion memory board and jumper configuration (except one) the instant memcheck hits the start of the mapped ram location it just stops with no error. Here is what I see:

```
***********************************************************
*                                                         *
*          rosco_m68k SysInfo & MemCheck utility          *
*        MC68010 CPU with Firmware 2.16 [RELEASE ]        *
* Firmware reports  1048576 bytes total contiguous memory *
*                                                         *
***********************************************************

Building memory map, please wait this may take a while...

Scanning for RAM: 1024K/14336K  

```

When I change the jumpers on the memory board the 1024k is replaced with the base address or (start of installed memory). 

If I disconnect the memory board or short all jumpers on the ram board then execution is successful because the memory board is being bypassed. With the exception of jumper pattern 0b111, the program will stop the instant it hits the mapped memory space. I even tried installing different memory configurations (even ones which aren't supported). For simplicity I'll give it as a four bit code where each bit represents a 1 megabyte even odd ram pair with the right most bit being base + $fffff and the left most bit being base + $300'000 pair. A zero means no installed and a one means installed. 

0b1111 -> hangs at memory start
0b0111 -> hangs at memory start
0b0011 -> hangs at memory start
0b0001 -> hangs at memory start
0b1110 -> hangs at base + $100'000
0b1100 -> hangs at base + $200'000
0b1000 -> hangs at base + $300'000
0b0000 -> no hang... but no memory installed.

I didn't try all combinations. I got the same result if JP4 is shorted or not as well. 

I also did an indepth inspection of mem board as well looking for solder bridges and found none. 
I am using Alliance AS6C4008-55PCN SRAM chips as well.

If I power it from an external 5v source or the FTDI cable the result is the same (obviously not at the same time).

There is something I'm obviously doing wrong but not sure what it is.

mifritscher, 2022-04-01:
I've the same problem. It seems to be a problem with memcheck, as e.g. peek&poke from ehbasic worked fine.

DrItanium, 2022-04-01:
I'm glad I'm not the only one. So what command did you run in ehbasic to test that (I don't know basic at all...)? I'd like to use it to validate that the memory expansion is working.

mifritscher, 2022-04-01:
Uh, I just used PRINT, PEEK and POKE from https://philpem.me.uk/leeedavison/68k/ehbasic/language.html.

something like
```
A = $<Address in HEX>
PRINT PEEK(A)
POKE (A,200)
PRINT PEEK(A)
```

should do.

DrItanium, 2022-04-01:
Thank you :D

For reference it is 

```
A = $<Address in HEX>
PRINT PEEK(A)
POKE A,200
PRINT PEEK(A)
```

Which worked for me. And the memory expansion board is working.

DrItanium, 2022-04-01:
I'm going to close this issue and open an new issue with a more descriptive title

closed 2022-04-01 by DrItanium