rosco-m68k/rosco_m68k #316: Memcheck hangs if memory expansion board is connected to a v2 based rosco_m68k board

DrItanium, 2022-04-01:
Running the memcheck example will hang during RAM scan if the memory expansion board is connected to the v2 classic board. 

The hang happens once the program hits the memory mapped area. It will not fail if the jumpers of the memory board are set to 0b111, no memory is installed in the expansion board board, or the board is not connected. 

I observed this on a 2.10 classic running firmware 2.16.

0xTJ, 2022-04-02:
I've replicated this on my board, with firmware 2.20.

0xTJ, 2022-04-02:
Huh, I think I've found the problem, it doesn't look like the code will advance the `current` pointer at all if the data was successfully read back. It will always loop forever as soon as it encounters valid expansion RAM. I should have a PR up shortly to fix this.

PR #317 (2022-04-04): Fix endless loop in memcheck with expansion RAM
This is a minor change where the scanning pointer `current` wasn't advanced when checking expansion RAM, leading to an infinite loop where the same address is checked forever.

Fixes #316.
closed 2022-04-04 by 0xTJ