rosco-m68k/rosco_m68k #330: Cannot produce pcb from gerbers

GBeushausen, 2022-06-05:
I try to get the PCB manufactured (R2). But the silkscreen is missing. If i export it myself with KiCad (don't know what i'm doing), the manufacturer says there are errors and cannot manufacture it. How can i get a working version of the PCB?

roscopeco, 2022-06-05:
Interesting. All the Gerbers for r2 are here: https://github.com/rosco-m68k/rosco_m68k/tree/develop/design/r2/CAMOutputs

The silkscreen for front and back are contained in the `rosco_m68k-F_SilkS.gbr` and `rosco_m68k-B_SilkS.gbr` files respectively. 

The board is four-layer, and the layer stack up for copper is:

* `rosco_m68k-Top.gbr` (Top)
* `rosco_m68k-Route2.gbr`
* `rosco_m68k-Route15.gbr`
* `rosco_m68k-Bottom.gbr` (Bottom)

The drill files can be found alongside the gerbers (`rosco_m68k-PTH.drl` and `rosco_m68k-NPTH.drl`).

For most fabs, it should be enough to simply zip all the files in the `CAMOutputs` directory and send that zip file. Some fabs will require you to assign a layer stack-up (which is as listed above).

Which manufacturer are you using? What are the specific errors they are complaining about?

GBeushausen, 2022-06-05:
Thanks for your help. I didn't used the develop branch. I used the Release2 branch and there, the silk screen was missing. With this branch i didn't get any errors, other than it was submitted as a 2 layer board and it seems to be a 4 layer board. I'm, using PCBWay.

roscopeco, 2022-06-05:
No probs. You had me slightly worried, but I just had a look through the release branches and the silkscreen appears to be there in all of them. The `release/version-2.0` branch is somewhat out of date though, if you were using a release branch it's always best to use the latest for board designs (so `release/version-2.2` in this case).

However, it's also fine to use `develop` in this instance as the board hasn't changed since the last release, glad to hear it worked for you 👍 

Okay, no worries, this should be fine for PCBWay :)

closed 2022-06-05 by roscopeco