rosco-m68k/rosco_m68k #338: BOM - What to buy?

GBeushausen, 2022-07-31:
I have got some boards produced for me. Now i want to build them, but i don't know what components exactly i have to buy.

Like in the BOM it says: 
4;"C7,C24,C13,C31,C12,C10,C32,C30,C29,C28,C9,C8,C3,C4,C5,C6,C27,C26,C25,C23,C22,C20";"C2.5-3";22;"100nF";;;
What does this mean? 100nf of what? Where to buy this?

Or this:
5;"R18,R17,R16,R15,R12,R8,R10,R32,R31,R9,R3,R7";"0207_10";12;"4K7";;;

What type of resistor is this?

Or this chip:
15;"IC6,IC2,IC5,IC3";"DIP-24_W7.62mm_Socket_LongPads";4;"ATF22V10C";;;

There are many subtypes of this chip, like 15pu, 20pi, 7x some say 10 macrocells, some 500 slpd. How do i know exactly which chip i need to buy?

roscopeco, 2022-07-31:
The resistors are all standard 0.25W axial through-hole resistors.

For the capacitors, `C2.5-3` are standard 2.5mm lead pitch radial non-polarised caps with the capacitance specified in the BOM (e.g. 100nF, 7pF, etc). I recommend MLCC. `E2,5-5` is a standard radial electrolytic polarised cap with 2.5mm lead pitch and the specified capacitance. `E3,5-10` (for the 470uF) is a standard radial electrolytic polarised cap with 3.5mm lead pitch. In other words, the letter (`C` or `E`) specifies whether this a ceramic or electrolytic cap, while the number immediately following specifies lead pitch and package size. All are through-hole.

For the ATF22V10C, I do not mandate a specific type as it depends to an extent what your programmer can handle. The board is tested with ATF22V10C-10PU , and ATF22V10C-7PX is known to work also.

roscopeco, 2022-08-06:
Again (https://github.com/rosco-m68k/rosco_m68k/issues/331), closing this as I assume it fixed your issue, though I can't really tell given no further feedback from yourself. I cannot begin to tell you how fulfilling I find your demands and our one-sided interactions, and how much I look forward to there being no more of them.

closed 2022-08-06 by roscopeco

GBeushausen, 2022-08-06:
@roscopeco I'm really thankful for your help. You just immediatly closed the issue after replying to me and i was not sure if i could still reply after you closed the issue.

roscopeco, 2022-08-06:
[image]