rosco-m68k/rosco_m68k #349: Error array subscript @ is outside array bounds

Berzerkula, 2023-07-04:
https://github.com/rosco-m68k/rosco_m68k/blob/a42a2f78be5507def89f6c46d2ec49db9581c695/code/software/libs/src/printf/include/printf.h#L59

I am using latest gcc and binutils for the m68k toolchain. Used -Wno-array-bounds  for the follwing:

m68k-elf-gcc -c -std=c11 -ffreestanding -ffunction-sections -fdata-sections -fomit-frame-pointer -Wall -Wextra -Werror -Wno-unused-function -pedantic -I../../../code/software/libs/build/include -mcpu=68010 -march=68010 -mtune=68010 -g -O2 -DROSCO_M68K  -o memcheck.o memcheck.c
In file included from ../../../code/software/libs/build/include/stdio.h:20,
                 from memcheck.c:16:
In function 'show_banner',
    inlined from 'main' at memcheck.c:400:3:
../../../code/software/libs/build/include/printf.h:59:16: error: array subscript 0 is outside array bounds of 'uint32_t[0]' {aka 'long unsigned int[]'} [-Werror=array-bounds=]
   59 | #define printf printf_
memcheck.c:346:5: note: in expansion of macro 'printf'
  346 |     printf("* Firmware reports %8ld bytes total contiguous memory *\n", *FW_MEMSIZE);
      |     ^~~~~~
In function 'main':
cc1: note: source object is likely at address zero
cc1: all warnings being treated as errors
make[1]: *** [../../../code/software/software.mk:109: memcheck.o] Error 1
make[1]: Leaving directory '/home/kb0iic/repos/rosco_m68k/code/software/memcheck'
make: *** [Makefile:31: memcheck] Error 2

0xTJ, 2023-07-05:
Is this an issue when building the current version in the `develop` branch? This issue, and the one in #348, *should* have already been fixed when we added GCC 13 support.

Berzerkula, 2023-07-05:
I do remember this involved a bit back, I'm seeing it in GCC 14 (using git for GCC and Binutils)  I did a make clean, git -fxd (really cleans it) and did a pull, and all is fine. Unsure how I saw it earlier. It is clear. I'll close this issue.

closed 2023-07-05 by Berzerkula