rosco-m68k/rosco_m68k #360: QUESTION: Is there anything special about IC3 (ATF22V10C) that you get in the kit?

blackdwarf, 2023-09-16:
I managed to boot up the computer (yay!) but I noticed that IC3, IC5, IC6, and the CPU get warm. I reseated the first 3 and tried to power it on again and no luck. Looking around what happened, I noticed that I managed to turn IC3 the wrong way. Curse my poor eyesight and poor lighting!

Anyways, DigiKey has these in stock, so wanted to buy a couple, but just wanted to check is there anything special I need to do with the replacement programming-wise? Or can I just plop in a new one (this time double checking if it is oriented right)?

Thanks!

roscopeco, 2023-09-16:
Hello!

Sorry to hear about this, but glad you've been able to get replacements.

The ATF22V10Cs are programmable logic devices, and will need to be programmed before use. You can find the JEDEC file you'll need to program IC3 here: https://raw.githubusercontent.com/rosco-m68k/rosco_m68k/develop/code/pld/r2/duartsel/ic3_duart_sel.jed

We recommend the TL866II+ programmer and the Minipro software - if you have those then you can also find a handy script to compile and burn the code for this chip in this directory: https://github.com/rosco-m68k/rosco_m68k/tree/develop/code/pld/r2/duartsel

If you don't have a programmer, please feel free to open a support case on Tindie - use either the "Contact Store" button on the main store page ( https://www.tindie.com/stores/rosco/ ) or there should be a similar button on your order details page to directly raise a question.

Thanks,
Ross

blackdwarf, 2023-09-16:
Thanks for the quick response, really appreciate it!

I do have a programmer, but it is a T48, which seems like it [doesn't work](https://gitlab.com/DavidGriffith/minipro/-/issues/270) with minipro. However, I will probably be able to just flash the PLD from the Windows software?

roscopeco, 2023-09-16:
No worries 🙂 Ah right, yeah, that's the reason I've still not gotten a T48 yet 😅 

I would think you should be able to flash the PLD fine from the Windows software using the JED file I linked above.

Please do update here if you have any issues with it 🙂

blackdwarf, 2023-09-18:
An update here (for posterity and other people who may be butter fingered like I was), and yet another question.

So, based on our discussion above, I took a chance and read IC3 in my T48. That worked, but the contents were definitely different than the JED file you linked. I blanked it, then wrote the JED file from the repo. It...worked! I can boot the board! I also compiled the BASIC from the repo, transferred it there and ran some code that worked. 

However, the CPU and the IC3, IC5, and IC6 do get warm. Is this normal? Or do I have another issue I need to look at? I am powering it off of an ordinary USB port.

roscopeco, 2023-09-19:
Awesome, thanks for the update, and glad to hear you’re up and running!

It’s totally normal for those chips to get warm when the system is running 🙂

blackdwarf, 2023-09-19:
Great! Then I'm good to go and try me some 68k stuff. Never had one when I was a kid. 😄

roscopeco, 2023-09-19:
Have fun! 🥳

closed 2023-09-19 by roscopeco