rosco-m68k/rosco_m68k #385: Null-terminated print traps should leave A1 pointing to the terminator

roscopeco, 2023-11-15:
In the interface reference docs, we state that the print traps leave A1 pointing to the null terminator.

This no longer appears to be the case, we save and restore that register in the trap code.

since this behaviour is useful, we should put it back as it was.

0xTJ, 2023-12-13:
Is there a one of the implementation that's known not to do as it's documented? I've checked most of the functions printing functions and those seemed to match what's documented.

`TRAP #14` uses `A0` while `TRAP #13` uses `A1`, so I could see if there was some confusion where the main system print traps save and restore `A1`. I might however have also just missed a spot where the behaviour is wrong.

roscopeco, 2024-04-07:
Closing this as superseded. If still an issue we can reopen or reraise 🙂

closed 2024-04-07 by roscopeco