rosco-m68k/rosco_m68k #457: can't compile develop branch

der-schmelle2, 2024-10-07:
Hello,
this is my first ‘issue’ so please be nice to me and give me hints if I did something wrong.
I am running macOS 15.0.1.
I have installed the toolchain and am now trying to compile the firmware in develop Branch. Unfortunately this does not work. See attached file with the information received from the terminal output.
Procedure for reproduction:
```
brew tap rosco-m68k/toolchain
brew install gcc-cross-m68k@13 vasm-m68k srecord minipro
brew install zstd
git clone https://github.com/rosco-m68k/rosco_m68k.git
git submodule update --init --recursive
git checkout master
make clean all
make burn
```
**** IS WORKING FINE ****

`git checkout develop`
M code/firmware/rosco_m68k_firmware/romfs/micropython/core
M code/firmware/rosco_m68k_firmware/romfs/mklittlefs
Switched to branch ‘develop’
our branch is up to date with ‘origin/develop’.
`make clean all`
*** SEE attached screen output ***
[error_develop_branch.txt](https://github.com/user-attachments/files/17275054/error_develop_branch.txt)

Please help.
Thanks Peter

der-schmelle2, 2024-10-07:
> Hello! 🙂
>
> Ah, you've caught us right in the middle of some large changes in develop,
> so there are some things that you'll need to do differently, but which
> aren't yet reflected in the documentation - those changes are in flight
> currently, sorry about that.
>
> The first thing you'll need to do is fetch submodules from git:
>
> git submodule update --init
>
> Once that's done, you'll need to be using our newer, more full-featured
> toolchain. It's currently pre-release, so will take a little while to
> install since it is built locally (it is HEAD-only until we cut a proper
> release). You can install it with:
>
> brew install --HEAD ***@***.***
>
> Once those are done, you should be able to built with make clean all as
> you did before.
>
> Hope that helps, let us know if you hit any more problems 🙂

Hello Ross,
Thank you very much for your message. I will carry out your suggested steps
tomorrow. I am really enthusiastic about your project. Thank you for that.

Best regards
Peter

roscopeco, 2024-10-08:
Hmm, the error `m68k-elf-objcopy: No such file or directory` suggests the old toolchain wasn't found.

There's a [PR currently in flight](https://github.com/rosco-m68k/rosco_m68k/pull/454) to remove the dependency, but in the meantime you need to have both installed.

You mentioned initially that you'd already installed the old toolchain - did you uninstall it?

der-schmelle2, 2024-10-08:
No, I have not uninstalled anything. Would it make sense to uninstall and
reinstall both?

roscopeco, 2024-10-08:
No, there shouldn't be any need to do that. The easiest thing to do while we're waiting to get this PR merged would be:

```
git checkout feature/remove-bare-metal-toolchain-dependency
```

and then I'll update here as soon as we get the PR merged and you can switch back to `develop` :)

roscopeco, 2024-10-09:
Update: the PR to remove the old toolchain dependency is merged, you should be able to build develop now if you pull.

der-schmelle2, 2024-10-09:
It's working fine. Thanks