rosco-m68k/rosco_m68k #460: Regression: `mcCheckDevice`, `mcReadDevice` appear non-functional for UART B

roscopeco, 2024-10-09:
When code is linked against Newlib `librosco`, `mcCheckDevice` and `mcReadDevice` appear non-functional for UART B.

Requires further testing, as very possible it's not limited to those two routines and/or UART B.

0xTJ, 2024-12-24:
Is there a specific program that can be used to replicate this? I tried modifying a `newlib` example, and explicitly linked `librosco`, but can't replicate this. On a rosco_m68k Classic V2 rev 2.2 (with the FTDI trace to UART B cut, so using an external FTDI) `mcCheckDevice` still returns `true` when a character is available, and I can correctly read in that character with `mcReadDevice`.

roscopeco, 2024-12-24:
> Is there a specific program that can be used to replicate this? I tried modifying a `newlib` example, and explicitly linked `librosco`, but can't replicate this. On a rosco_m68k Classic V2 rev 2.2 [...]

Thanks for looking at this @0xTJ ! 

I noted this down when testing other things and it’s very possible I had other “moving parts” in play at the time. If you’re not able to replicate then that puts my mind at rest, I’m happy to close this on that basis and if it shows up again we can revisit.

0xTJ, 2024-12-24:
Sounds good, closing for now! :)

closed 2024-12-24 by 0xTJ